Brandawg93/PeaNUT #186: [Bug]: v5.4.0 throwing errors in the log

vladaurosh, 2025-02-19:
### Describe the bug

Latest v5.4.0 is throwing this error in the log:
```
[auth][error] MissingSecret: Please define a `secret`. Read more at https://errors.authjs.dev#missingsecret
peanut  |     at /.next/server/src/middleware.js:380:53307
peanut  |     at sd (/.next/server/src/middleware.js:380:55929)
peanut  |     at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
```

App seems to be working fine. I see that v5.4.0 introduced authentication but it doesn't look to be mandatory. 

### Debug Output

N/A

### Steps to reproduce

1.N/A

### Device Type

Rpi

### UPS Device

Eaton 5P

vladaurosh, 2025-02-19:
Just did compose down and up but issue persists.
```
peanut  | Killed
peanut  | Generated new AUTH_SECRET
peanut  | 
peanut  | > peanut@5.4.0 start
peanut  | > HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js
peanut  | 
peanut  |    ▲ Next.js 15.1.7
peanut  |    - Local:        http://localhost:8080
peanut  |    - Network:      http://0.0.0.0:8080
peanut  | 
peanut  |  ✓ Starting...
peanut  |  ✓ Ready in 282ms
peanut  | [auth][error] MissingSecret: Please define a `secret`. Read more at https://errors.authjs.dev#missingsecret
peanut  |     at /.next/server/src/middleware.js:380:53307
peanut  |     at sd (/.next/server/src/middleware.js:380:55929)
peanut  |     at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
```

vladaurosh, 2025-02-19:
Btw, just tried `npx auth secret` in container's cli, but it killed process and restarted container. 
```
npm warn exec The following package was not found and will be installed: auth@1.2.3
Killed
```

Brandawg93, 2025-02-19:
Try the latest test and let me know if it's fixed. `brandawg93/peanut:test`. If so, I'll push a quick fix.

vladaurosh, 2025-02-19:
Looks good now.
`Generated new AUTH_SECRET: bQuGJY3A9jbyobBGEUUB22bL08cWlMt+6QNMDgIIHwo=`

Brandawg93, 2025-02-19:
Awesome. Thanks for testing! v5.4.1 will be up soon.

closed 2025-02-19 by Brandawg93