Brandawg93/PeaNUT #187: [Feature Request]: Allow using docker secrets to set values in the settings file

velvetpolygons, 2025-02-22:
### Enhancement of issue

I would like to be able to configure PeaNUT by using docker secrets to avoid having password in clear text in settings file.

### Solution

Homepage use {{HOMEPAGE_VAR_X}} and {{HOMEPAGE_FILE_X}} and replace them in their config files.
https://gethomepage.dev/installation/docker/#using-environment-secrets
https://github.com/gethomepage/homepage/blob/dev/src/utils/config/config.js

I think a similar solution to it could work.
Settings:
```yaml
NUT_SERVERS:
  - HOST: {{PEANUT_FILE_HOST}}
    PORT: {{PEANUT_FILE_PORT}}
    USERNAME: {{PEANUT_FILE_USERNAME}}
    PASSWORD: {{PEANUT_FILE_PASSWORD}}
```
Docker compose:
```yaml
peanut:
    image: brandawg93/peanut:latest
    container_name: PeaNUT
    ports:
      - 8080:8080
    volumes:
      - /path/to/config:/config
    environment:
      - WEB_PORT=8080
      - PEANUT_FILE_HOST=/run/secrets/peanut-host
      - PEANUT_FILE_PORT=/run/secrets/peanut-port
      - PEANUT_FILE_USERNAME=/run/secrets/peanut-username
      - PEANUT_FILE_PASSWORD=/run/secrets/peanut-password
    secrets:
      - peanut-host
      - peanut-port
      - peanut-username
      - peanut-password
```
_This works with multiple servers. Just add different environment variables for each server._

### Alternatives

Configuring instantlinux/nut-upsd, I can set a secret as follows:
```yaml
services:
  nut-upsd:
    image: instantlinux/nut-upsd
    container_name: nut-upsd
    environment:
      - DRIVER=usbhid-ups
    devices:
      - /dev/bus/usb:/dev/bus/usb
    secrets:
      - nut-upsd-password
```
Inside a bash script they read content of the file and sets the value of API_PASSWORD to it.
Later creates their configuration file from environment variables.
https://gitlab.com/instantlinux/docker-tools/-/blob/main/images/nut-upsd/entrypoint.sh

_Could be harder to apply here if you wish to have more than one server._

Brandawg93, 2025-03-06:
I have a new test version with #200 that should add compatibility for docker secrets using the `PEANUT_` prefix. Try it out and let me know if it works for you.

`brandawg93/peanut:test`

velvetpolygons, 2025-03-08:
Great. I have now tested and it seems to works. However, I found a problem/bug. If I go to the settings page and load the yaml configuration it doesn't load the file from disk, it has the version with replaced values, showing everything replaced. If I now press the save button, it overwrites settings.yml on disk with the version with replaced values.
If I try changing it on the page back to:
```yaml
NUT_SERVERS:
  - HOST: nut-upsd
    PORT: 3493
    USERNAME: upsmon
    PASSWORD: {{PEANUT_PASSWORD}}
```
I get the following on disk:
```yaml
NUT_SERVERS:
  - HOST: nut-upsd
    PORT: 3493
    USERNAME: upsmon
    PASSWORD:
      '[object Object]': null
```
Also saw Manage Server have the replaced values. Not sure if this is an easy fix or not. Might be a much bigger change than what I thought considering the changes needed to the GUI.

Brandawg93, 2025-03-08:
Oh that is true. I'll see what I can do and get back to you.