Brandawg93/PeaNUT #190: [Bug]: An error occurred in the Server Components render.

JdCpuWiz, 2025-02-24:
### Describe the bug

This is the first time i have tried to set up peanut.  On previous installs I have used NUTGui, but PeaNUT has integration with HomePage so I wanted to try it
When trying to add a new UPS (the first one) when clicking the test button I am getting an error message.

[image]

Docker Compose:
services:  
  peanut:
    image: brandawg93/peanut:latest
    container_name: peanut
    environment:
      - PUID=1000
      - PGID=1000
      - TZ=America/Chicago
      - WEB_PORT=8080
      - AUTH_SECRET=Q5l7jWd+CMO2yhhLsTKeTT4myZlM1WPM8m+hYnGRU9c=
    restart: unless-stopped
    volumes:
      - /home/shad/docker/peanut/config:/config
    devices:
      - /dev/bus/usb
    ports:
      - 8080:8080
    networks:
      - nut-ups
networks:
  nut-ups:
    external: true

[image]

### Debug Output

from docker container log:

> peanut@5.4.2 start
> HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js

   ▲ Next.js 15.1.7
   - Local:        http://localhost:8080
   - Network:      http://0.0.0.0:8080

 ✓ Starting...
AUTH_SECRET already exists: Q5l7jWd+CMO2yhhLsTKeTT4myZlM1WPM8m+hYnGRU9c=
 ✓ Ready in 107ms
 ⨯ Error: Login failed
    at R (.next/server/chunks/2385.js:1:4262) {
  digest: '1146865488'
}

### Steps to reproduce

1.  I just configured the docker compose file and started the container.
2. Navigated to the IP of the server at port 8080 and the WEB GUI worked.
3. I tried to enter the details of the UPS and recieved the error message.

### Device Type

Proxmox LXC container running Ubuntu 24.10LTS

### UPS Device

Cyberpower UPS

Brandawg93, 2025-02-24:
Thanks for the heads up. It definitely should tell you the error instead of saying that, but I see your specific issue.

> Error: Login failed

JdCpuWiz, 2025-02-24:
So my issue is just a failed login?

JdCpuWiz, 2025-02-25:
I just pulled v5.3.0 and ran the identical docker compose file against that version and it loaded without issue.

Brandawg93, 2025-02-25:
Correct. v5.3.0 was not actually checking the authentication. It was a bug. You could also remove the username and password from the NUT server config in your settings and it should work.

aroundmyroom, 2025-02-25:
Just updated and indeed when the pwd is incorrect you get this red message, when it is correct it is doing the test (y'll not see the result, also no error' when applying a 2nd time 'connection successful' is displayed. It looks a little different than when I was testing it from the test repository

Brandawg93, 2025-02-25:
I just created a new test version with #195 that should fix this. Let me know if this is better.

`brandawg93/peanut:test`

aroundmyroom, 2025-02-25:
pulling a test works fine in my dockge .

closed 2025-02-25 by Brandawg93

JdCpuWiz, 2025-02-25:
working for me as well!!  Thank you!  Great work.