Brandawg93/PeaNUT #193: [Bug]: Dopdown list missing for mutliple UPS devices

fcr, 2025-02-24:
### Describe the bug

There is no drop down list for selecting which of multiple UPS devices to monitor. I only manage to switch devices by constantly refreshing the page in my browser, but this is unreliable. I have two devices at different addresses defined and both are recognised by Peanut.

See, no dropdown box or any other means of selecting a device:
[image]

Or have I missed something?

### Debug Output

None

### Steps to reproduce

1. Add two devices to settings and, test and apply
2. Open browser on Peanut

### Device Type

Tested with Firefox, Safari on MacOS, iOS and PadOS, for Peanut version 5.4.2 running in Docker on a Synology NAS.

### UPS Device

Eaton 850i and Eaton 650i

Brandawg93, 2025-02-25:
[It's the values in brackets](https://github.com/Brandawg93/PeaNUT/blob/main/config/ups.conf)

fcr, 2025-02-25:
Thanks. That probably explains it. 
I only have a `settings.yml` file in the `/config` folder.
i didn't realise that I have to provide a 'ups.config' file.

Brandawg93, 2025-02-25:
PeaNUT doesn't need the file. Your NUT server needs it.

fcr, 2025-02-25:
Hmm. I have two nut servers - each a Synology NAS. Each one with an Eaton  UPS. 
So each NUT server would only know about one UPS.
Any way to work around this?

fcr, 2025-02-25:
.Any way to work around this other than have separate Peanut instances for each server, that is

Brandawg93, 2025-02-25:
This may help.

https://github.com/Brandawg93/PeaNUT/issues/184#issuecomment-2666977302

fcr, 2025-02-25:
Great. Thanks. I'll give it a try.
As far as I am concerned, you can close this issue.
And thanks for a great little utility.

Brandawg93, 2025-02-25:
Ok. If you run into any other issues, feel free to reach out.