Brandawg93/PeaNUT #194: [Bug]: [AggregateError: All promises were rejected] when using API

My-Random-Thoughts, 2025-02-25:
### Describe the bug

When trying the API  `/api/v1/devices/{device}` the above error message is displayed twice in the event log.
That is, one API call, two error messages.

The `{device}` value I am trying is the "default", `ups` as well as other values.

### Debug Output

    2025-02-25T11:22:40.205365286Z > peanut@5.4.2 start
    2025-02-25T11:22:40.205367741Z > HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js
    2025-02-25T11:22:40.205369152Z 
    2025-02-25T11:22:40.478376936Z    ▲ Next.js 15.1.7
    2025-02-25T11:22:40.478494981Z    - Local:        http://localhost:8080
    2025-02-25T11:22:40.478587584Z    - Network:      http://0.0.0.0:8080
    2025-02-25T11:22:40.478603063Z 
    2025-02-25T11:22:40.478702791Z  ✓ Starting...
    2025-02-25T11:22:40.582158853Z Generated new AUTH_SECRET: xxx
    2025-02-25T11:22:40.582498693Z  ✓ Ready in 134ms

    2025-02-25T11:25:31.549256828Z  ⨯ [AggregateError: All promises were rejected]
    2025-02-25T11:25:31.549444918Z  ⨯ [AggregateError: All promises were rejected]

    Code  Details
    500     Undocumented

    Error: Internal Server Error
    Response headers

    connection: keep-alive
    date: Tue,25 Feb 2025 11:25:31 GMT
    keep-alive: timeout=5
    transfer-encoding: chunked
    vary: RSC,Next-Router-State-Tree,Next-Router-Prefetch,Next-Router-Segment-Prefetch 

### Steps to reproduce

SETTINGS.YML:

    NUT_SERVERS:
      - HOST: 192.168.42.6
        PORT: 3493
        USERNAME: observer
        PASSWORD: password
    INFLUX_HOST: ''
    INFLUX_TOKEN: ''
    INFLUX_ORG: ''
    INFLUX_BUCKET: ''
    INFLUX_INTERVAL: 10

DOCKER COMPOSE:

    name: peanut
    services:
        peanut:
            image: brandawg93/peanut:latest
            container_name: peanut
            restart: unless-stopped
            network_mode: bridge

            ports:
                - 82:8080

            volumes:
                - /home/user/peanut:/config

            environment:
                - TZ=Europe/London
                - WEB_PORT=8080

            deploy:
                resources:
                    limits:
                        cpus: "1"
                        memory: 128m

### Device Type

Docker

### UPS Device

APC

### Last Working Version

None, initial usage

Brandawg93, 2025-02-25:
Thanks for the report! It's definitely an issue and I can reproduce it. But it does mean that your NUT server can't find the device associated with the value you put into `{device}`. It should return 404 instead of 500.

Brandawg93, 2025-02-25:
I just created a new test version with #195 that should fix this. Let me know if this is better.

`brandawg93/peanut:test`

Brandawg93, 2025-02-25:
Great! This is now live in v5.5.0.

closed 2025-02-25 by Brandawg93