Brandawg93/PeaNUT #197: [Bug]: The latest version does not recognize the Cyberpower UPS connected to the Synology NAS

tOOmki, 2025-03-03:
### Describe the bug

The latest version does not recognize the Cyberpower UPS connected to the Synology NAS. 
The APC UPS is correctly detected. 
I updated from an older version (I don't remember which one, maybe even v4).

### Debug Output

.

### Steps to reproduce

1. 

### Device Type

Proxmox

### UPS Device

APC, Cyberpower

### Last Working Version

v4

Brandawg93, 2025-04-07:
Do you still have this issue? And if so, does [this](https://github.com/Brandawg93/PeaNUT/wiki/Synology-Configuration) help you?

tOOmki, 2025-04-08:
This is probably a different error than I thought.
This is a result of the lack of individual names on the main page with the list of devices.

Even though I have configured 1x APC, 2x Cyberpower, only "ups" is displayed on the list and that's the first one I added.

Brandawg93, 2025-04-09:
Gotcha. That definitely sounds related to #184. If it turns out to be something different, just let me know.

closed 2025-04-09 by Brandawg93