Brandawg93/PeaNUT #198: [Bug]: Control menu disappeared after upgrading from v5.2.1 to v5.3.0+

WhistleMaster, 2025-03-03:
### Describe the bug

After upgrading from v5.2.1 to v5.3.0 (and in the latest v5.5.0), the dropdown menu that contained options for "Run quick test", "Run deep test", "Disable beeper", "Enable beeper", and "Mute beeper" has disappeared from the UI. This menu was previously accessible from the top right corner of the interface.

### Debug Output

N/A

### Steps to reproduce

1. Install and run Peanut v5.2.1
2. Note the presence of the beeper control menu in the top right corner
3. Upgrade to v5.3.0 or later (v5.5.0)
4. Note that the beeper control menu is no longer accessible

### Device Type

Proxmox / Debian VM / Docker - Peanut

### UPS Device

Back-UPS RS 1500G

### Last Working Version

v5.2.1 (12.02.2025)

Brandawg93, 2025-03-03:
This would be expected if you did not have a username and password defined in your NUT server settings in PeaNUT, or if your username and password were incorrect. You may want to double check those values.

WhistleMaster, 2025-03-04:
Thank you for your response. I've tested, this isn't related to authentication. I've tested with v5.5.0 using a username and password and successfully logged in, but the beeper control menu (with options for "Run quick test", "Run deep test", "Disable beeper", etc.) still doesn't appear in the top right corner.

Brandawg93, 2025-04-09:
There are two different authentication areas. One for peanut and one for each NUT server. You may want to recheck the ladder one. If you continue to have issues, just let me know.