Brandawg93/PeaNUT #205: [Bug]: Application hangs if multiple UPSs are defined and one is non responsive

mx5gr, 2025-03-14:
### Describe the bug

Hello,

I am using the latest version, 5.6.0, and I have defined two UPSs, let's call them A and B.

Due to a hardware fault, the 1st UPS (A) has been temporarily decommissioned, the second (B) works as usual.

When launching PeaNUT, we get: "Application error: a client-side exception has occurred while loading myserver.com (see the browser console for more information)"

### Debug Output

Console output (Mozilla Firefox) is:

```
Uncaught TypeError: e is undefined
    NextJS 62

r
        getValue
        getValue
        es
        lS
        ot
        ov
        uf
        uc
        uc
        ue
        uQ
        uI
        uV
        uR
        uR
        i7
        l$
        lH
        i
        n
        r
        r
        i
        r
        setTimeout handler*l
        r
        batch
        #s
        setData
        onSuccess
        f
        promise callback*b
        start
        fetch
        #M
        onSubscribe
        subscribe
        x
        lH
        ob
        oG
        oX
        oG
        oX
        oG
        oX
        oG
        oX
        oG
        oX
        oG
        oX
        oG
        oX
        oG
        oX
        oG
        oX
        oG
        oX
        oG
```

### Steps to reproduce

1. Define 2 UPS for monitoring within config
2. Pull out USB cable to the 1st one, with remote NUT server (the one with the USB attached) still running
3. Launch PeaNUT

### Device Type

Raspi 4

### UPS Device

Eaton

xxbiohazrdxx, 2025-03-25:
Also happening to me

Brandawg93, 2025-03-26:
I have a new test version with #212 that should fix this. Could someone try it and let me know?

`brandawg93/peanut:test`

Brandawg93, 2025-03-28:
This should be fixed in v5.6.1.

closed 2025-03-28 by Brandawg93