Brandawg93/PeaNUT #218: Is it possible to set a 24 clock display via docker-compose?

pmhausen, 2025-03-29:
Hello,

I use this docker-compose YAML to deploy PeaNUT from Dockge:
```
services:
  peanut:
    image: brandawg93/peanut:latest
    container_name: PeaNUT
    restart: unless-stopped
    ports:
      - 15001:8080
    volumes:
      - /mnt/nvme/apps/peanut:/config
    environment:
      - WEB_PORT=8080
networks: {}
x-dockge:
  urls:
    - http://peanut.ettlingen.hausen.com:15001
```

Timezone of Europe/Berlin seems to be inherited correctly from the host but the web UI always shows an AM/PM value. I'd prefer 24h.

Is that possibly somehow? I tried various permutations of locale settings via environment variables to no avail.

Thanks!
Patrick

Brandawg93, 2025-03-29:
Looks like it was forced to 12 hour clock. #219

This should fix it but I can't seem to figure out how to change my mac's time preferences. 😅

pmhausen, 2025-03-30:
How can I test it? Could you add a tag I can pull in Docker?

Brandawg93, 2025-03-30:
Sure. I just made a test version. You can try it via `brandawg93/peanut:test`.

pmhausen, 2025-04-01:
When I switch "latest" to "test" I still get version 5.6.1. Also I cannot see a "test" tag in the repo.

Brandawg93, 2025-04-01:
Correct. The version will stay the same. The test will show on docker hub and not the repo.

https://hub.docker.com/repository/docker/brandawg93/peanut/tags/test/sha256-ba42ab3a151fa2d0033205d6c80102a150ddcd194bf064f1008a0431b9cd4713

pmhausen, 2025-04-01:
Still showing 07:07:32 PM - guess I have to mess with the environment (LANG and friends) again.

Brandawg93, 2025-04-01:
I think it's a browser setting not an environment setting. I couldn't get mine to change either. I could force the clock to show 24 hour with an environment variable, but I'm pretty sure there is a way to default to it on your machine. But I can't seem to find it for my Mac.

pmhausen, 2025-04-01:
My Mac OS and Safari definitely default to 24h clock. If you have more important stuff to do, feel free to close it. If you like an environment or YAML setting would of course be great.

Brandawg93, 2025-04-01:
All good! It's a unique ask that I've never thought about before so I like the challenge. 😄 

Try pulling the test tag again and this time just click on the "Last Updated" footer at the bottom of the page.

pmhausen, 2025-04-04:
Works for the "last updated" display but not (yet) for the time in the voltage graph.

Brandawg93, 2025-04-05:
Got it. The latest test should work now. You just need to click the time at the bottom and refresh the page.

pmhausen, 2025-04-06:
To quote User Friendly's Pitr: da - now workink.

Is that setting persisted anywhere? I have an out of container volume for the YAML configuration but it seems not to end up there.

Kind regards,
Patrick

Brandawg93, 2025-04-06:
It's persisted in your browser. So you may have to do that in different browsers. I may look into making it an official setting later. But glad it's working now! I'll release it in the next official version.

closed 2025-04-07 by Brandawg93