Brandawg93/PeaNUT #220: [Bug]: Missing InfluxDB values when ON BATTERY

ce3a, 2025-04-02:
### Describe the bug

PeaNUT not writing "input.voltage" to InfluxDB when the UPS is in ON BATTERY status (ie. when input.voltage is 0).

[image]

[image]

### Debug Output

```
ERROR: Write to InfluxDB failed. Error [HttpError]: failure writing points to database: partial write: field type conflict: input field "input.voltage" on measurement "ups1" is type string, already exists as type float dropped=1
    at <unknown> (HttpError: failure writing points to database: partial write: field type conflict: input field "input.voltage" on measurement "ups1" is type string, already exists as type float dropped=1)
    at IncomingMessage.<anonymous> (.next/server/chunks/8397.js:4:5031) {
  statusCode: 422,
  statusMessage: 'Unprocessable Entity',
  body: '{"code":"unprocessable entity","message":"failure writing points to database: partial write: field type conflict: input field \\"input.voltage\\" on measurement \\"ups1\\" is type string, already exists as type float dropped=1"}',
  contentType: 'application/json; charset=utf-8',
  headers: [Object],
  json: [Object],
  code: 'unprocessable entity',
  _retryAfter: 0
}
Error [HttpError]: failure writing points to database: partial write: field type conflict: input field "input.voltage" on measurement "ups1" is type string, already exists as type float dropped=1
    at <unknown> (HttpError: failure writing points to database: partial write: field type conflict: input field "input.voltage" on measurement "ups1" is type string, already exists as type float dropped=1)
    at IncomingMessage.<anonymous> (.next/server/chunks/8397.js:4:5031) {
  statusCode: 422,
  statusMessage: 'Unprocessable Entity',
  body: '{"code":"unprocessable entity","message":"failure writing points to database: partial write: field type conflict: input field \\"input.voltage\\" on measurement \\"ups1\\" is type string, already exists as type float dropped=1"}',
  contentType: 'application/json; charset=utf-8',
  headers: [Object],
  json: [Object],
  code: 'unprocessable entity',
  _retryAfter: 0
}
```

### Steps to reproduce

1. Configure logging data to InfluxDB
2. Let UPS enter ON BATTERY status
3. Check input.voltage in InfluxDB bucket

### Device Type

x86 Proxmox

### UPS Device

CyberPower OR600ERM1U

Brandawg93, 2025-04-07:
The error is a bug, but the reason it's not writing makes sense. When your UPS is on battery, you don't have an input voltage. The NUT server is probably returning "N/A". I'm working on #223 to handle the errors better, but the issue will persist since it's an expected outcome.

Brandawg93, 2025-04-07:
This should now be better handled in v5.7.0. Feel free to reopen if you have any other issues.

ce3a, 2025-04-07:
There seems to be a similar error in **v5.7.0**. (Note: I don't have permissions to reopen this issue.)

```
> peanut@5.7.0 start
> HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js
   ▲ Next.js 15.2.4
   - Local:        http://localhost:8080
   - Network:      http://0.0.0.0:8080
 ✓ Starting...
 ▲ [next-ws] has started the WebSocket server
Generated new AUTH_SECRET: xxx
 ✓ Ready in 116ms
[Error: Failed to find Server Action "001a80c1ecc2d93671992847cd4679f285e58b9a0d". This request might be from an older or newer deployment. Original error: Cannot read properties of undefined (reading 'workers')
Read more: https://nextjs.org/docs/messages/failed-to-find-server-action]
ERROR: Write to InfluxDB failed. Error [HttpError]: failure writing points to database: partial write: field type conflict: input field "input.voltage" on measurement "ups1" is type string, already exists as type float dropped=1
    at <unknown> (HttpError: failure writing points to database: partial write: field type conflict: input field "input.voltage" on measurement "ups1" is type string, already exists as type float dropped=1)
    at IncomingMessage.<anonymous> (.next/server/chunks/7787.js:15:5031) {
  statusCode: 422,
  statusMessage: 'Unprocessable Entity',
  body: '{"code":"unprocessable entity","message":"failure writing points to database: partial write: field type conflict: input field \\"input.voltage\\" on measurement \\"ups1\\" is type string, already exists as type float dropped=1"}',
  contentType: 'application/json; charset=utf-8',
  headers: [Object],
  json: [Object],
  code: 'unprocessable entity',
  _retryAfter: 0
}
```
When the UPS is on battery, upsc is returning `input.voltage: 0.0`. 

```
$ upsc ups1@pve1 | grep -E "(ups.status|input.voltage|output.voltage):"
Init SSL without certificate database
input.voltage: 229.0
output.voltage: 229.0
ups.status: OL

$ upsc ups1@pve1 | grep -E "(ups.status|input.voltage|output.voltage):"
Init SSL without certificate database
input.voltage: 0.0
output.voltage: 230.0
ups.status: OB DISCHRG
```

I believe PeaNUT should also write `input.voltage: 0.0` to influxdb in this case. This would make visualizing the data much easier.

If there is anything I could do to help debugging or testing, please let me know.

ce3a, 2025-04-07:
I believe the issue is here https://github.com/Brandawg93/PeaNUT/blob/v5.7.0/src/server/nut.ts#L193.

```ts
vars[key] = { value: num || value, description }
```
Numeric values like "0.0" from the UPS are stored as strings due to the use of  `(num || value)` that treats **0 as falsy**. Hence, `"230.0"` is stored as float, but `"0.0"` is stored as string.

Maybe it should be replaced with something like this?

```ts
vars[key] = { value: isNaN(num) ? value : num, description }
```

Brandawg93, 2025-04-08:
That's a great idea! I think it's actually more simple than that. I've created a test version with #227. Let me know if that works for you.

Brandawg93, 2025-04-09:
Perfect! I'll release it in the next version.

Brandawg93, 2025-04-09:
The fix has been released in v5.7.1.

closed 2025-04-09 by Brandawg93