Brandawg93/PeaNUT #229: [Bug]: Terminal does not work?

aroundmyroom, 2025-04-10:
### Describe the bug

when starting the terminal feature I see following:

{"type":"error","message":"Unauthorized"}

So dunno why I would be getting a json style error message, but something is wrong ;)

help what is advertised does not work.

### Debug Output

Running the peaNUT in Proxmox with LXC without docker

### Steps to reproduce

1. 

### Device Type

Proxmox LCX

### UPS Device

APC

TypicalZedF, 2025-04-11:
I have this same error, i'm not exactly sure why

Brandawg93, 2025-04-14:
#230

aroundmyroom, 2025-04-15:
OK, 
be noticed that the terminal is white when there is no dark mode 
I am able to enter some commands .. 

HELP
Commands: HELP VER GET LIST SET INSTCMD LOGIN LOGOUT USERNAME PASSWORD STARTTLS
VER
Network UPS Tools upsd 2.8.0 - http://www.networkupstools.org/

But further than that I have no idea what is the usage ;)

But now no error anymore ..

Brandawg93, 2025-04-15:
Great! Sounds like it's working as expected. I'll release a new version shortly.

Brandawg93, 2025-04-15:
Released in v5.7.2

closed 2025-04-15 by Brandawg93