Brandawg93/PeaNUT #237: [Bug]: Fails to build with next 15.3.0

Spartan-II-117, 2025-04-26:
### Describe the bug

I attempted to update an LXC container and got the following debug output, traced it to the PNPM run build command

I was using the script from https://community-scripts.github.io/ProxmoxVE/scripts?id=peanut
if this should be a bug elsewhere, please let me know

### Debug Output

root@peanut:/opt/peanut# $STD pnpm run build

> peanut@5.7.2 build /opt/peanut
> next build

   ▲ Next.js 15.3.0
   - Experiments (use with caution):
     ✓ reactCompiler

   Creating an optimized production build ...
 ✓ Compiled successfully in 10.0s

Failed to compile.

./src/client/components/connect.tsx
76:18  Error: Replace `absolute·left-0·top-0·flex·h-full·w-full·flex-col·items-center·justify-center·bg-background` with `bg-background·absolute·top-0·left-0·flex·h-full·w-full·flex-col·items-center·justify-center`  prettier/prettier
86:24  Error: Replace `relative·flex·flex-row·justify-around·overflow-hidden·border·border-border·bg-card·` with `border-border·bg-card·relative·flex·flex-row·justify-around·overflow-hidden·border·`  prettier/prettier
87:26  Error: Replace `w-full·max-w-sm·bg-card` with `bg-card·w-full·max-w-sm`  prettier/prettier
96:26  Error: Replace `w-full·border-border-card·bg-background` with `border-border-card·bg-background·w-full`  prettier/prettier
108:26  Error: Replace `w-full·border-border-card·bg-background` with `border-border-card·bg-background·w-full`  prettier/prettier
121:26  Error: Replace `w-full·border-border-card·bg-background·` with `border-border-card·bg-background·w-full·`  prettier/prettier
133:28  Error: Replace `z-10·rounded-r-none·border-r-0·border-border-card·bg-background` with `border-border-card·bg-background·z-10·rounded-r-none·border-r-0`  prettier/prettier
140:28  Error: Replace `relative·overflow-hidden·rounded-l-none·border·border-l-0·border-border-card·bg-background` with `border-border-card·bg-background·relative·overflow-hidden·rounded-l-none·border·border-l-0`  prettier/prettier

info  - Need to disable some ESLint rules? Learn more here: https://nextjs.org/docs/app/api-reference/config/eslint#disabling-rules
 ELIFECYCLE  Command failed with exit code 1.

### Steps to reproduce

1. 

### Device Type

LXC container on proxmox

### UPS Device

N/A

Brandawg93, 2025-05-02:
I've updated a few packages in the latest release. It should fix this issue. If not, just let me know.

Spartan-II-117, 2025-05-02:
problems persist in 5.7.3. here's my results:

please let me know if there's anything i can do to help track this down

root@peanut:/opt/peanut# pnpm i
Lockfile is up to date, resolution step is skipped
Already up to date

> peanut@5.7.3 prepare /opt/peanut
> husky && next-ws patch --yes

.git can't be found[next-ws] Next.js v15.3.1 is not officially supported, a maximum of v15.2.4 is recommended.
[next-ws] Proceeding with the latest patch
[next-ws] If you encounter any issues please report them at https://github.com/apteryxxyz/next-ws/issues
[next-ws] Patching Next.js v15.3.1 with '>=13.5.1 <=15.2.4'
[next-ws] ✔ Add WebSocket server setup script to NextNodeServer constructor.
[next-ws] ✔ Prevent Next.js from immediately closing WebSocket connections
[next-ws] ✔ Add SOCKET type to Next types
[next-ws] Saving patch trace file...
[next-ws] All done!
Done in 1.5s
root@peanut:/opt/peanut# pnpm run build

> peanut@5.7.3 build /opt/peanut
> next build

   ▲ Next.js 15.3.1
   - Experiments (use with caution):
     ✓ reactCompiler

   Creating an optimized production build ...
 ✓ Compiled successfully in 12.0s

Failed to compile.

./src/client/components/connect.tsx
76:18  Error: Replace `absolute·left-0·top-0·flex·h-full·w-full·flex-col·items-center·justify-center·bg-background` with `bg-background·absolute·top-0·left-0·flex·h-full·w-full·flex-col·items-center·justify-center`  prettier/prettier
86:24  Error: Replace `relative·flex·flex-row·justify-around·overflow-hidden·border·border-border·bg-card·` with `border-border·bg-card·relative·flex·flex-row·justify-around·overflow-hidden·border·`  prettier/prettier
87:26  Error: Replace `w-full·max-w-sm·bg-card` with `bg-card·w-full·max-w-sm`  prettier/prettier
96:26  Error: Replace `w-full·border-border-card·bg-background` with `border-border-card·bg-background·w-full`  prettier/prettier
108:26  Error: Replace `w-full·border-border-card·bg-background` with `border-border-card·bg-background·w-full`  prettier/prettier
121:26  Error: Replace `w-full·border-border-card·bg-background·` with `border-border-card·bg-background·w-full·`  prettier/prettier
133:28  Error: Replace `z-10·rounded-r-none·border-r-0·border-border-card·bg-background` with `border-border-card·bg-background·z-10·rounded-r-none·border-r-0`  prettier/prettier
140:28  Error: Replace `relative·overflow-hidden·rounded-l-none·border·border-l-0·border-border-card·bg-background` with `border-border-card·bg-background·relative·overflow-hidden·rounded-l-none·border·border-l-0`  prettier/prettier

info  - Need to disable some ESLint rules? Learn more here: https://nextjs.org/docs/app/api-reference/config/eslint#disabling-rules
 ELIFECYCLE  Command failed with exit code 1.

Brandawg93, 2025-05-05:
Hm. Honestly, I'm not really sure what's going on. Looks to be a simple linting error that could be ignored. You could probably modify the files once downloaded by the script. The link in your error you attached shows you how.

Brandawg93, 2025-09-28:
Close as this is stale and there are many new versions since this.

closed 2025-09-28 by Brandawg93