Brandawg93/PeaNUT #245: [Bug]:

chrisp250, 2025-05-07:
### Describe the bug

When trying to set up the connection to nut, the test feature shows it can't connect to the UPS, however if I apply the settings I can see all the live data on the dashboard.

Nut server is running as an add-in in Home Assistant. I'm able to connect to it from other clients without any issues.

Thank you

### Debug Output

PeaNUT  | 
PeaNUT  | > peanut@5.7.4 start
PeaNUT  | > HOSTNAME=$WEB_HOST PORT=$WEB_PORT node server.js
PeaNUT  | 
PeaNUT  |    ▲ Next.js 15.3.1
PeaNUT  |    - Local:        http://localhost:8080
PeaNUT  |    - Network:      http://0.0.0.0:8080
PeaNUT  | 
PeaNUT  |  ✓ Starting...
PeaNUT  |  ▲ [next-ws] has started the WebSocket server
PeaNUT  |  ✓ Ready in 117ms
PeaNUT  | Generated new AUTH_SECRET: xxxxxxxxxxxxxxxxxxxxxxx
PeaNUT  |  ⨯ Error: Invalid credentials
PeaNUT  |     at o (.next/server/chunks/ssr/_4127a837._.js:420:1360) {
PeaNUT  |   digest: '3599929065'
PeaNUT  | }

### Steps to reproduce

1. Set up a new server from *Manage Servers*
2. Use the *Test* button

### Device Type

Docker container inside a Proxmox VM

### UPS Device

APC BE7700

### Last Working Version

Just installed it for the first time

Brandawg93, 2025-05-07:
This should be fixed in the latest test version. Let me know if it works for you.

`brandawg93/peanut:test`

Brandawg93, 2025-05-07:
This should now be fixed in v5.7.5. Let me know if you continue to have issues.

chrisp250, 2025-05-07:
> This should now be fixed in v5.7.5. Let me know if you continue to have issues.

Hi there,
Thank you for looking into this. I updated the container to v5.7.5 but the test function still doesn't work. The software still works fine otherwise.

Cheers
Christian

Brandawg93, 2025-05-08:
Ah. Then it's probably something with your NUT server credentials. `Error: Invalid credentials` It'll actually still work without the credentials. You just can't run commands.

chrisp250, 2025-05-08:
> Ah. Then it's probably something with your NUT server credentials. `Error: Invalid credentials` It'll actually still work without the credentials. You just can't run commands.

I understand, however I don't see how that could be possible. I'm copying and pasting the credentials from the server into peanut. Also I can connect to the server from another client using this same credentials and I'm able to change parameters using `upsrw`.

I feeling I'm missing something here. Could this be an encrypted vs plain text password sort of issue?

Thank you for your help
Christian

chrisp250, 2025-05-08:
I got it to work by adding the option:
`upsmon: master`

to the user I had created. 
You might want to add in your wiki that the user requires this option. 
Thank you

stevestebo1987, 2025-05-09:
where did you put the upsmon: master?

chrisp250, 2025-05-09:
> where did you put the upsmon: master? I am having the same issue where if i change battery.low...... it does not give me access denied but does not change the value. Is that the name of your UPS too or username?

I have two UPS, one connected to **Home Assistant** and the other to a **Raspberri PI** running RPI OS.
This is my Home Assistant plugin config:
```
users:
  - username: nutter
    password: xxxxxxx
    instcmds:
      - all
    actions:
      - set
    upsmon: master
devices:
  - name: serversups
    driver: usbhid-ups
    port: auto
    config:
      - desc = "Datacentre UPS"
mode: netserver
shutdown_host: false
log_level: debug
```

This is my PI config for `upsd.users`:
```
[nutter]
	password = xxxxxxxxx
	actions = SET
	actions = fsd
	instcmds = ALL
	upsmon master
```

Brandawg93, 2025-05-12:
After looking into this, [`master` is actually deprecated for `primary`](https://networkupstools.org/docs/developer-guide.chunked/net-protocol.html). The config uses primary instead. It could be that old NUT servers still rely on `master`.

closed 2025-05-12 by Brandawg93