Brandawg93/PeaNUT #246: [Bug]: Can't change any config that has a pencil next to it

stevestebo1987, 2025-05-09:
### Describe the bug

I can't change any config that has a pencil next to it.  After putting the SET permissions under my user I don't receive the error not allowed, but now it just does not change the value.  Am I am able to update the values from the webpage?

### Debug Output

Value does not change, not sure how to get any more logging.  My dozzle logging app is not showing anything that I can go off of to troubleshoot on my own.

### Steps to reproduce

1. I have all the credentials correct, because I am able to run upsrw on the nuts docker and it changes there.

### Device Type

Docker Image running on Linux Debian VM

### UPS Device

APC, default usbhid-ups

### Last Working Version

Just installed

stevestebo1987, 2025-05-09:
By the way it does work with the API calls under the API doc, it just does not work when you use the pencil edit on the main page.

Brandawg93, 2025-06-02:
Is this still an issue? And do you see errors in your console?

stevestebo1987, 2025-06-02:
I can try again, but a couple of days ago it still wouldn’t change.  I could change it thru the API but not the main page where the edit buttons are.Thank you,Steven Smith

stevestebo1987, 2025-06-02:
That’s the weird thing.  I don’t see errors, it just switches back to the original setting.Thank you,Steven Smith

ee-boi, 2025-06-06:
I have the same problem, when I go to change a setting with a pencil mark next to it and click the green check mark to save it I get the following response` Invalid response: ERR ACCESS-DENIED ` 

I am running the containers similar to the guide [here](https://github.com/Brandawg93/PeaNUT/wiki/Simple-Default-Configuration) with the only change being the port number on the web ui. I do not see any messages in the docker logs.

Brandawg93, 2026-05-21:
Please try with v6 and let me know if the issue still persists.

@ee-boi if you do not see a pencil, then peanut is not seeing that it is editable.