Brandawg93/PeaNUT #260: [Bug]: 24-hour format

HanaO00, 2025-06-29:
### Describe the bug

Hello 24h format doesn't seem to work

[image]

### Debug Output

Conf file

TIME_FORMAT: 24-hour

### Steps to reproduce

1. 

### Device Type

PC

### UPS Device

Eaton / USB

Brandawg93, 2025-06-29:
Thanks for submitting a bug! Can you give me steps to reproduce this? Everything seems to work for me when I change it via the settings page.

HanaO00, 2025-06-29:
Of course, I installed via Docker, part of the compose concerned

`peanut:
    image: brandawg93/peanut:latest
    container_name: peanut
    environment:
      - NUT_HOST=xxx
      - NUT_PORT=3493
      - WEB_PORT=xxx
    volumes:
      - ./config:/config
    ports:
      - xxx:80xxx88
    restart: unless-stopped`

Then 

Settings -> General -> 24-hour -> Apply

HanaO00, 2025-06-29:
Of course, I installed via Docker, part of the compose concerned

```
image: brandawg93/peanut:latest
container_name: peanut
environment:
  - NUT_HOST=xxx
  - NUT_PORT=3493
  - WEB_PORT=xxx
volumes:
  - ./config:/config
ports:
  - xxx:xxx
restart: unless-stopped
```

Then

Settings -> General -> 24-hour -> Apply

Brandawg93, 2025-07-06:
I'm still unable to reproduce this. What browser and version are you using? It could be related to that.

HanaO00, 2025-07-10:
Firefox and Vanadium, the two will display this, I tried to put the timezone variable but the problem persists

Brandawg93, 2025-07-10:
I was able to reproduce this on Firefox but only with a production build and not with a dev build, meaning that it's most likely a NextJS issue with Firefox. 

I created a new test build with an updated version of NextJS that may fix the issue. Feel free to try it with `brandawg93/peanut:test` and let me know.

HanaO00, 2025-07-11:
It still doesn't work, neither on Firefox nor on Vanadium with the test version

The problem probably comes from my browsers i don't know