Brandawg93/PeaNUT #267: [Bug]: Interface Crashes and instructs user to look at browser Console

sdjernes, 2025-07-12:
### Describe the bug

Hello,

I have two CyberPower UPSes that are connected to my NUT server via SNMP.  These two UPSes do not report UPS status as of Nut 2.8.0 on Debian 12 / ProxMox 8.4.  This causes on the PeaNUT home screen to show an "N/A" in the status field.  When you click the details button it proceeds to load a white page saying "Look at browser console for error".  I have put the console output in the debug section of this form.

I am glad to provide any more details that I can as I like the nice simple user interface of PeaNUT and I think my family will be able to use it to see power stats when it works.  

I have one usb-hid CyberPower UPS that is working correctly so I have seen what this can do. 

I am working on building a copy of NUT 2.8.3 for Debian 12 so will see if anything changes.

Thank You,
Shawn

### Debug Output

192.168.1.57/:1 The resource http://192.168.1.57:3000/_next/static/css/d2ed3b33afe94554.css was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
192.168.1.57/:1 The resource http://192.168.1.57:3000/_next/static/css/f6342a102a4fd40b.css was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
page-df8bef53abcb13ac.js:1 Uncaught TypeError: e.startsWith is not a function
    at eV (page-df8bef53abcb13ac.js:1:38714)
    at eK (page-df8bef53abcb13ac.js:1:44903)
    at l2 (103adca0-8fb3d4cb49cf9509.js:1:50099)
    at ox (103adca0-8fb3d4cb49cf9509.js:1:69890)
    at oU (103adca0-8fb3d4cb49cf9509.js:1:81085)
    at ic (103adca0-8fb3d4cb49cf9509.js:1:112390)
    at 103adca0-8fb3d4cb49cf9509.js:1:112235
    at is (103adca0-8fb3d4cb49cf9509.js:1:112243)
    at u5 (103adca0-8fb3d4cb49cf9509.js:1:109326)
    at iV (103adca0-8fb3d4cb49cf9509.js:1:130104)
    at iR (103adca0-8fb3d4cb49cf9509.js:1:128323)
    at iU (103adca0-8fb3d4cb49cf9509.js:1:128675)
    at 103adca0-8fb3d4cb49cf9509.js:1:127952
eV @ page-df8bef53abcb13ac.js:1
eK @ page-df8bef53abcb13ac.js:1
l2 @ 103adca0-8fb3d4cb49cf9509.js:1
ox @ 103adca0-8fb3d4cb49cf9509.js:1
oU @ 103adca0-8fb3d4cb49cf9509.js:1
ic @ 103adca0-8fb3d4cb49cf9509.js:1
(anonymous) @ 103adca0-8fb3d4cb49cf9509.js:1
is @ 103adca0-8fb3d4cb49cf9509.js:1
u5 @ 103adca0-8fb3d4cb49cf9509.js:1
iV @ 103adca0-8fb3d4cb49cf9509.js:1
iR @ 103adca0-8fb3d4cb49cf9509.js:1
iU @ 103adca0-8fb3d4cb49cf9509.js:1
(anonymous) @ 103adca0-8fb3d4cb49cf9509.js:1
setTimeout
i @ 9125-1a53c949ec852aa6.js:4
s @ 9125-1a53c949ec852aa6.js:4
batch @ 9125-1a53c949ec852aa6.js:4
#A @ 9125-1a53c949ec852aa6.js:4
setData @ 9125-1a53c949ec852aa6.js:4
onSuccess @ 9125-1a53c949ec852aa6.js:4
f @ 9125-1a53c949ec852aa6.js:4
Promise.then
b @ 9125-1a53c949ec852aa6.js:4
start @ 9125-1a53c949ec852aa6.js:4
fetch @ 9125-1a53c949ec852aa6.js:4
#f @ 9125-1a53c949ec852aa6.js:1
onSubscribe @ 9125-1a53c949ec852aa6.js:1
subscribe @ 9125-1a53c949ec852aa6.js:4
(anonymous) @ 9125-1a53c949ec852aa6.js:1
ad @ 103adca0-8fb3d4cb49cf9509.js:1
oq @ 103adca0-8fb3d4cb49cf9509.js:1
uh @ 103adca0-8fb3d4cb49cf9509.js:1
um @ 103adca0-8fb3d4cb49cf9509.js:1
uh @ 103adca0-8fb3d4cb49cf9509.js:1
um @ 103adca0-8fb3d4cb49cf9509.js:1
uh @ 103adca0-8fb3d4cb49cf9509.js:1
um @ 103adca0-8fb3d4cb49cf9509.js:1
[... 139 lines cut ...]
um @ 103adca0-8fb3d4cb49cf9509.js:1
uh @ 103adca0-8fb3d4cb49cf9509.js:1
iS @ 103adca0-8fb3d4cb49cf9509.js:1
(anonymous) @ 103adca0-8fb3d4cb49cf9509.js:1
w @ 2841-97bac3974c629e19.js:1
cyber-sddesk:1 The resource http://192.168.1.57:3000/_next/static/css/d2ed3b33afe94554.css was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
cyber-sddesk:1 The resource http://192.168.1.57:3000/_next/static/css/f6342a102a4fd40b.css was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.

### Steps to reproduce

1. Open the PeaNUT page
2. Click the details button next to a CyberPower SNMP UPS in the list
3. 

### Device Type

ProxMox LXC Container

### UPS Device

CP1500PFCLCDa w/ RMCARD205 fw v1.4.4

### Last Working Version

none

Brandawg93, 2025-07-12:
Thanks for the bug report. I made a fix in the latest test version. Could you try it out and let me know if it works for you?
`brandawg93/peanut:test`

Brandawg93, 2025-07-16:
The fix has been published in v5.9.0. Let me know if you still have issues.