Brandawg93/PeaNUT #283: [Feature Request]: batteryRuntime column in homepage view

nixta, 2025-08-07:
### Enhancement of issue

The status I see on the overview grid is great, but I'd like to see how much runtime each UPS estimates it might provide under the current load. I'm not very familiar with UPS behavior and management, so perhaps there's a nuanced reason this isn't considered a reliable or sensible thing to show, but since the dashboard shows the battery charge and the current load, this feels like it should be OK?

### Solution

Add a Battery Remaining column after Current Load in the homepage devices table.

Brandawg93, 2025-08-07:
https://github.com/Brandawg93/PeaNUT/blob/b534a1fd57625f1e482b675c22b1747718646a44/src/client/components/device-wrapper.tsx#L267

Most ups devices report this value. It may be that yours does not so it is hidden.

Brandawg93, 2025-08-08:
I've created a new test version that shows runtime in a tooltip when you hover over the battery charge cell. Try it out and let me know what you think.

`brandawg93/peanut:test`

#284

nixta, 2025-08-08:
Looks great. Have left a comment on the PR. Thank you! That was very quick :)

Brandawg93, 2025-08-08:
This is now live in v5.11.0.