Brandawg93/PeaNUT #287: [Feature Request]: Work with macosx-ups NUT driver

nixta, 2025-08-10:
### Enhancement of issue

I think there may be an issue with the `macosx-ups` NUT driver. I have an iMac, and had to give up trying to manage the UPS via the usual `usbhid-ups` driver (permission issues), so have it connected via the `macosx-ups` driver. I can connect to it with an `nc <my-mac> 3493` and issue a `LIST UPS` call to get a response, but when I try to add the Mac as a NUT server, Peanut tells me there's a connection issue and to check the logs (though it doesn't tell me which logs, and I'm having trouble finding them 😬).

Is Peanut maybe not happy with the handshake with the server when it's just got a single `macosx-ups` driver to report? How can I dig up the logs to find out?

Thanks!

### Solution

I'd like to be able to see my Cyberpower UPS that is connected to my iMac.

Brandawg93, 2025-08-10:
They should be in the docker container logs. 

`docker logs`

nixta, 2025-08-11:
Thanks!

I'm running it as a proxmox LXC container. Will have to do some digging to learn where that means the logs go.

Either way, would you expect what I'm doing to work?

nixta, 2025-08-11:
Ok. Thanks. I will keep churning at it then. For reference, here's the driver: https://networkupstools.org/docs/man/macosx-ups.html

It has very limited arguments compared with the [usbhid-ups driver](https://networkupstools.org/docs/man/usbhid-ups.html), so I was wondering if there's some way that might trip up PeaNUT's New Connection logic. Is that a possible avenue for me to explore (if you happen to know off the top of your head)?

Brandawg93, 2025-08-13:
Oh. Duh. You're right. You can also do the same by directly connecting to the upsd application that is connected to the driver. It's similar to how the backend of PeaNUT works.

https://networkupstools.org/docs/developer-guide.chunked/net-protocol.html

nixta, 2025-08-16:
Thanks! Sorry - been a busy week. Will try to make time for this this weekend.

nixta, 2025-08-31:
I haven't forgotten about this - unfortunately we're just renovating so the machine and UPS in question have been put away for a little while! Sorry!!

closed 2026-05-21 by Brandawg93