Brandawg93/PeaNUT #295: [Feature Request]: Add device_id to /api/v1/devices

superstrom, 2025-08-29:
### Enhancement of issue

Peanut 5.13.0 via docker

Unless I am missing something obvious, the /api/v1/devices result does not include the device_id that is necessary to call the rest of the apis.  Is there a separate api to get the device ids?

For example, this is the full result of calling /api/v1/devices for my setup (don't judge how old my devices are):

```
[
  {
    "battery.charge": 100,
    "battery.charge.low": "10",
    "battery.charge.warning": 50,
    "battery.date": "2001/09/25",
    "battery.mfr.date": "2019/09/18",
    "battery.runtime": 1436,
    "battery.runtime.low": "120",
    "battery.type": "PbAc",
    "battery.voltage": 13.7,
    "battery.voltage.nominal": 12,
    "device.mfr": "American Power Conversion",
    "device.model": "Back-UPS NS 675M1",
    "device.serial": "4B1938P14531",
    "device.type": "ups",
    "driver.debug": 0,
    "driver.flag.allow_killpower": 0,
    "driver.name": "usbhid-ups",
    "driver.parameter.interrupt_pipe_no_events_tolerance": -1,
    "driver.parameter.pollfreq": 30,
    "driver.parameter.pollinterval": 2,
    "driver.parameter.port": "auto",
    "driver.parameter.serial": "4B1938P14531",
    "driver.parameter.synchronous": "auto",
    "driver.parameter.vendorid": "051d",
    "driver.state": "quiet",
    "driver.version": "2.8.3",
    "driver.version.data": "APC HID 0.100",
    "driver.version.internal": 0.62,
    "driver.version.usb": "libusb-1.0.28 (API: 0x0100010A)",
    "input.sensitivity": "medium",
    "input.transfer.high": "139",
    "input.transfer.low": "92",
    "input.transfer.reason": "input voltage out of range",
    "input.voltage": 117,
    "input.voltage.nominal": 120,
    "ups.beeper.status": "disabled",
    "ups.delay.shutdown": "20",
    "ups.firmware": "930.a8 .D",
    "ups.firmware.aux": "a8",
    "ups.load": 12,
    "ups.mfr": "American Power Conversion",
    "ups.mfr.date": "2019/09/18",
    "ups.model": "Back-UPS NS 675M1",
    "ups.productid": 2,
    "ups.realpower.nominal": 360,
    "ups.serial": "4B1938P14531",
    "ups.status": "OL",
    "ups.test.result": "No test initiated",
    "ups.timer.reboot": 0,
    "ups.timer.shutdown": -1,
    "ups.vendorid": "051d"
  },
  {
    "battery.charge": 100,
    "battery.charge.low": "10",
    "battery.charge.warning": 50,
    "battery.date": "2001/09/25",
    "battery.mfr.date": "2016/09/01",
    "battery.runtime": 584,
    "battery.runtime.low": "120",
    "battery.type": "PbAc",
    "battery.voltage": 27.2,
    "battery.voltage.nominal": 24,
    "device.mfr": "American Power Conversion",
    "device.model": "Back-UPS RS1000G",
    "device.serial": "3B1635X16644",
    "device.type": "ups",
    "driver.debug": 0,
    "driver.flag.allow_killpower": 0,
    "driver.name": "usbhid-ups",
    "driver.parameter.interrupt_pipe_no_events_tolerance": -1,
    "driver.parameter.pollfreq": 30,
    "driver.parameter.pollinterval": 2,
    "driver.parameter.port": "auto",
    "driver.parameter.serial": "3B1635X16644",
    "driver.parameter.synchronous": "auto",
    "driver.parameter.vendorid": "051d",
    "driver.state": "quiet",
    "driver.version": "2.8.3",
    "driver.version.data": "APC HID 0.100",
    "driver.version.internal": 0.62,
    "driver.version.usb": "libusb-1.0.28 (API: 0x0100010A)",
    "input.sensitivity": "medium",
    "input.transfer.high": "147",
    "input.transfer.low": "88",
    "input.transfer.reason": "input voltage out of range",
    "input.voltage": 116,
    "input.voltage.nominal": 120,
    "ups.beeper.status": "disabled",
    "ups.delay.shutdown": "20",
    "ups.firmware": "868.L4 -P.D",
    "ups.firmware.aux": "L4 -P",
    "ups.load": 25,
    "ups.mfr": "American Power Conversion",
    "ups.mfr.date": "2016/09/01",
    "ups.model": "Back-UPS RS1000G",
    "ups.productid": 2,
    "ups.realpower.nominal": 600,
    "ups.serial": "3B1635X16644",
    "ups.status": "OL",
    "ups.test.result": "No test initiated",
    "ups.timer.reboot": 0,
    "ups.timer.shutdown": -1,
    "ups.vendorid": "051d"
  }
]
```

### Solution

Add device_id to each object returned by api/v1/devices

Brandawg93, 2025-08-29:
The IDs are typically defined by you in the ups.conf file of your NUT config.

https://github.com/Brandawg93/PeaNUT/blob/bd3973192c497024a112de7f378005b44fb5cac2/config/ups.conf#L1

It some instances, though, (QNAP, Synology), they are predefined and you may have to find that file to see them.

superstrom, 2025-08-30:
This is perfect! Thanks for the quick implementation.